SymbiFlow/yosys #17: Yosys cannot infer BRAM correctly on current master+wip

mkurc-ant, 2019-03-12:
Yosys on current master+wip (SHA: 22eaab1) cannot infer BRAM (or any memory) when given a verilog array with a series of `initial` statements.

I've checked that the last correctly working commit is 98104b3.

In order to reporoduce:
1. Clone the Yosys repo (master+wip branch)
2. Build it
3. Set env. variable `YOSYS` to point to the Yosys binaty
4. Clode the symbiflow-arch-defs repo (master 2e7d736) and run `make env`
5. Build EBLIF file for `tests/9-soc/picosoc`
6. Check how many BRAMS are there in the synthesized design. There should be 2 instead of 1.

I don't know if it is an upstream bug or not so I am posting it here.

mkurc-ant, 2019-03-12:
This PR in YosysHQ introduced the unwanted behavior: https://github.com/YosysHQ/yosys/pull/843

mkurc-ant, 2019-03-12:
Created an issue upstream: https://github.com/YosysHQ/yosys/issues/867

mkurc-ant, 2019-03-14:
So the problem is now fixed on YosysHQ (f0b2d8e). @litghost Can you integrate `master+wip` from YosysHQ again? Or at least changes from https://github.com/YosysHQ/yosys/pull/868

litghost, 2019-03-14:
> So the problem is now fixed on YosysHQ ([f0b2d8e](https://github.com/SymbiFlow/yosys/commit/f0b2d8e467998876ad2cc14232d30ff7892982a3)). @litghost Can you integrate `master+wip` from YosysHQ again? Or at least changes from [YosysHQ#868](https://github.com/YosysHQ/yosys/pull/868)

I can, but FYI you could have also made the PR.

mkurc-ant, 2019-03-14:
@litghost Ok, I've created a PR: https://github.com/SymbiFlow/yosys/pull/19

closed 2019-03-14 by litghost