SymbiFlow/yosys #27: Full support for attributes in Yosys

mkurc-ant, 2019-05-28:
Support for attributes on parameters was successfully landed (https://github.com/SymbiFlow/yosys/pull/25). However, there are several cases of attribute usage still not supported by the Yosys.

- [ ] Module declaration: `(* optimize *) module control ( ... );`
- [ ] Module instance: `my_module (* my_attribute *) my_module_instance (...);`
- [ ] Variable (reg/wire): `(* state_variable *) reg [7:0] S;`
- [ ] Port connection: `dff il (q, d, (* clock_line *) ckl, rst);`
- [ ] Suffix of an operator: `sum = a + (* ripple_adder *) b;`
- [ ] Function call: `assign next_state = fsm_func(* meally_fsm *) (state, a, b, c);`

mkurc-ant, 2019-05-28:
@mithro I created this issue to keep track of attribute related features we want in Yosys. Those examples are just a subset of all possible attribute placement cases.

Anyway, can you give an real world example of usage of an attribute on a port connection ?

mithro, 2019-05-29:
This is how I wanted to use the attribute on a port connection;
```verilog
	MUX2 rmux (
		(* pack = "LUT5toLUT6; LUT5toLUT7; LUT5toLUT8" *)
		.I0(n5lut_0_out),
		(* pack = "LUT5x2" *)
		.I1(n5lut_1_out),
		.S(A6),
		.O(f6mux_O)
	);
```

mkurc-ant, 2019-05-30:
I started to work on attributes on port connections. And I encountered a problem - it seems that at some point Yosys swaps what RTLIL::SigSpec object refers to but not its attributes (which I added).

mithro, 2019-05-31:
These examples are from [Verilog — 2001 - A Guide to the New Features of the Verilog® Hardware Description Language](https://link.springer.com/book/10.1007/978-1-4615-1713-9) Section [`25. Attributes`](https://link.springer.com/chapter/10.1007%2F978-1-4615-1713-9_27),

```verilog
always @(state) //state machine with one-hot encoding
  (* full_case, parallel_case *) case (state)
  3'b001: next_state = 3'b010;
  3'b010: next_state = 3'b100;
  3'b100: next_state = 3'b001;
endcase 
```

> The attribute is specified as a prefix to a declaration, a module item, a statement or
a port connection. Three examples are:
```verilog
(* optimize *) module control ( ... );

(* state_variable *) reg [7:0] S;

dff il (q, d, (* clock_line *) ckl, rst};
```

> The attribute is specified as a suffix to an operator or the function name in a function call. 
```verilog
sum = a + (* ripple_adder *) b;

assign next_state = fsm_func(* meally_fsm *) (state, a, b, c); 
```

mkurc-ant, 2019-05-31:
@mithro It's working: https://github.com/SymbiFlow/yosys/pull/30