SymbiFlow/yosys #41: [Branch] attributes-on-parameters

mithro, 2019-09-05:
### Why did we need this? (what does this change enable us to do)

Enables Verilog attributes to be used on parameters.

### What did it change?

The Yosys RTIL to support attributes on parameters.

### Should it be merged upstream - if not, when can we delete it?

Unknown.

### More Info

See https://github.com/SymbiFlow/yosys/issues/27

litghost, 2019-09-05:
I believe this branch can be deleted, and I thought I had.

mithro, 2019-09-05:
@litghost - Upstream Yosys only has the ability to ignore attributes on parameters? This is full attribute on parameters support with tests...

litghost, 2019-09-05:
> @litghost - Upstream Yosys only has the ability to ignore attributes on parameters? This is full attribute on parameters support with tests...

I do not know, but I believe upstream yosys implemented another way to hold attributes on parameters?

eddiehung, 2019-09-06:
`read_verilog -pwires` preserves parameter values as constant driven wires. I'm not sure if it moves param attributes onto those wires, if not, can be easily fixed.

Can you provide more justification about what this enables, and/or a use case? Perfect round trip with Verilog in and Verilog out is not, in general, something synthesis tools can be expected to be good at since their purpose in life is to perform transformations.

mkurc-ant, 2019-09-17:
I managed to modify the verilog backend to output all "parameters as wires" properly as verilog `parameter` statements.  The option can be enabled in the verilog backend by specifying the `-pwires2params` switch. I've also added support for writing string parameter values as strings (not as integers) and real values as numbers.

There is also a test for verilog round trip which runs a verilog file twice through Yosys. Then it compares whether outputs after 1st and 2nd pass are equal.

I've created a new branch with this change: https://github.com/SymbiFlow/yosys/commits/attributes-on-parameters2