SymbiFlow/yosys #42: attributes-on-parameters branch currently broken

mithro, 2019-09-05:
When trying to rebase onto `upstream/master` and create the `attributes-on-parameters` branch I seem to have broken things;
```
backends/json/json.cc:119:21: error: use of undeclared identifier 'attributes'
                for (auto &attr : attributes) {
                                  ^
backends/json/json.cc:121:70: error: use of undeclared identifier 'param'
                        f << stringf("        %s%s: ", for_module ? "" : "    ", get_name(param.first).c_str());
                                                                                          ^
backends/json/json.cc:122:26: error: use of undeclared identifier 'param'
                        write_parameter_value(param.second);
                                              ^
backends/json/json.cc:116:53: warning: unused parameter 'parameters' [-Wunused-parameter]
        void write_parameters(const dict<IdString, Const> &parameters, bool for_module=false)
                                                           ^
backends/json/json.cc:140:3: error: use of undeclared identifier 'write_attributes'
                write_attributes(module->attributes, /*for_module=*/true);
                ^
backends/json/json.cc:144:3: error: use of undeclared identifier 'write_module_parameters'; did you mean 'write_parameters'?
                write_module_parameters(module);
                ^~~~~~~~~~~~~~~~~~~~~~~
                write_parameters
backends/json/json.cc:116:7: note: 'write_parameters' declared here
        void write_parameters(const dict<IdString, Const> &parameters, bool for_module=false)
             ^
backends/json/json.cc:144:27: error: reference to type 'const dict<Yosys::RTLIL::IdString, Yosys::RTLIL::Const>' could not bind to an lvalue of type 'Yosys::RTLIL::Module *'
                write_module_parameters(module);
                                        ^~~~~~
backends/json/json.cc:116:53: note: passing argument to parameter 'parameters' here
        void write_parameters(const dict<IdString, Const> &parameters, bool for_module=false)
                                                           ^
backends/json/json.cc:183:4: error: use of undeclared identifier 'write_attributes'
                        write_attributes(c->parameters);
                        ^
backends/json/json.cc:186:4: error: use of undeclared identifier 'write_attributes'
                        write_attributes(c->attributes);
                        ^
backends/json/json.cc:228:4: error: use of undeclared identifier 'write_attributes'
                        write_attributes(w->attributes);
                        ^
1 warning and 9 errors generated.
make: *** [Makefile:613: backends/json/json.o] Error 1
make: *** Waiting for unfinished jobs....
```

Don't have time to look at things now, but probably an easy fix

mithro, 2019-09-05:
@mkurc-ant I probably won't get to this until tomorrow, if you get a chance PTAL and fix the branch.

mkurc-ant, 2019-09-05:
@mithro I can fix that. But why would we want to use this code? The solution implemented here was not accepted by the Yosys devs. The upstream Yosys already has support for attributes on parameters - they are mocked by dummy wires.

mkurc-ant, 2019-09-06:
@mithro Now I see. We have merged this implementation of attributes on parameters into our master+wip but it was not accepted and not merged upstream.

If we want to merge with upstream we need to remove this code as it will cause conflict.

mkurc-ant, 2019-09-06:
@mithro I fixed the branch by reverting the merge of https://github.com/YosysHQ/yosys/pull/1109. It builds but the conflict with upstream master remains.