SymbiFlow/yosys #60: Branch : quicklogic : Issue while loading the edf file on spde "$_dlatch_p_.edi' "

kkumar23, 2020-03-06:
## Steps to reproduce the issue
Synthesize the attached design with yosys for the quicklogic.
Now load the edf onto spde tool, Spde throws the below error:

Error    : nm_1007: Library 'C:\QuickLogic\QuickWorks_2016.2_Release\spde\data\PolarPro-III\edif\$_dlatch_p_.edi'  is not supported for this family

It looks like the library of yosys is missing the dlatch [implementation.](url
[rtl.zip](https://github.com/SymbiFlow/yosys/files/4298296/rtl.zip)

[uart_yosys.zip](https://github.com/SymbiFlow/yosys/files/4298299/uart_yosys.zip)
[uart_spde.zip](https://github.com/SymbiFlow/yosys/files/4298302/uart_spde.zip)

)

kkumar23, 2020-03-18:
Any update on this issue?

glatosinski, 2020-03-24:
We've added support for latches for Quicklogic branch. However, the designs you've introduced won't work, since there is no inout support in Yosys for Quicklogic. I'm working on inout support in Yosys now.

glatosinski, 2020-03-24:
The support for inout in Quicklogic's script in Yosys is added. Your design should now be parsed correctly by SpDE.

kkumar23, 2020-03-27:
Hi Glatosinski, 
   I used the latest branch https://github.com/antmicro/yosys.git -b quicklogic quicklogic-yosys. to compile the Yosys. and updated the script "pip3 install git+https://github.com/antmicro/yosys-SpDE-flow.git" , i still get the same latch error in spde.

glatosinski, 2020-03-27:
The branch was updated yesterday's afternoon, sorry for the delay. Please check if it works for you. I've downloaded now the latest version of Yosys from https://github.com/antmicro/yosys.git (quicklogic branch) and it worked with SpDE.

kkumar23, 2020-03-27:
Thank you, I am able to run now.

closed 2020-03-31 by glatosinski