SymbiFlow/yosys #61: Branch: Quicklogic : Missing FIFO ports ( in ram8k_2x1_cell_macro) in the EDIF generated by Yosys

rakeshm75, 2020-03-09:
## Steps to reproduce the issue
1. Synthesize the attached design using yosys 
2. Convert Yosys EDIF file to SpDE complaint format
3. Load the design in SpDE
4. SpDE throws up the errors 

This design is using both the RAM's and FIFO's. The EDF generated by Yosys has missing FIFO ports in the RAM/ FIFO definition (ram8k_2x1_cell_macro). The following ports are missing:
                      1. Almost_Empty
                      2. Almost_Full
                      3. PUSH_FLAG
                      4. POP_FLAG
I have attached the design here (rtl and edf).
[Test_Design_1.zip](https://github.com/SymbiFlow/yosys/files/4305228/Test_Design_1.zip)

glatosinski, 2020-03-12:
After changes related to #62 and #63 it seems that the ports are preserved in EDIF file, so now the problem should disappear. The changes included fixes in cleaning routines and adding `keep` directive for SpDE-related blackboxes. Use https://github.com/antmicro/yosys/tree/quicklogic (`quicklogic` branch).

rakeshm75, 2020-03-16:
The issue is fixed. So closing the issue.

Thanks & Best Regards,
Rakesh

closed 2020-03-16 by rakeshm75