SymbiFlow/yosys #62: Branch: Quicklogic : Yosys optimizes the design completely when no IO's/ PADs defined in the design

rakeshm75, 2020-03-09:
## Steps to reproduce the issue
1. Synthesize any design without IO pads using yosys

IP designs on S3 devices can be used for processing data within S3, without external IO defined in the IP design. S3 has ASSP block (M4) which interfaces with the FPGA block, so there could be design where M4 sends data to FPGA, FPGA IP (like FFT) process the data and sends it back to M4. No external IO, in the FPGA IP, need to be defined in these scenario.

But Yosys, completely optimizes the design when it does not see any IO defined in the FPGA design.
[Test_Design2.zip](https://github.com/SymbiFlow/yosys/files/4305352/Test_Design2.zip)

glatosinski, 2020-03-12:
I updated https://github.com/antmicro/yosys/tree/quicklogic fork, where the support for QuickLogic devices currently resides. Now the `qlal4s3b_cell_macro` module has attribute `keep` that will preserve this cell, and communication with it.

rakeshm75, 2020-03-16:
The issue is fixed. So closing the issue.

Thanks & Best Regards,
Rakesh

closed 2020-03-16 by rakeshm75