SymbiFlow/yosys #63: Branch: Quicklogic : Yosys optimizes the design completely

rakeshm75, 2020-03-09:
## Steps to reproduce the issue
1. Synthesize the attached design using yosys
2. Convert Yosys EDIF file to SpDE complaint format
Yosys completely optimizes the design but when we enable the debug pins then it synthesizes the design correctly.
3. Un-comment //`define USE_DEBUG_PORT in the AL4S3B_FPGA_Top.v file, we are enabling the debug pins
4. Synthesize the attached design using yosys
Now Yosys synthesizes the design correctly.

Attached the design.
[Test_Design3.zip](https://github.com/SymbiFlow/yosys/files/4305498/Test_Design3.zip)

glatosinski, 2020-03-12:
I updated https://github.com/antmicro/yosys/tree/quicklogic fork, where the support for QuickLogic devices currently resides. Now the `qlal4s3b_cell_macro` module has attribute `keep` that will preserve this cell, and communication with it.

rakeshm75, 2020-03-16:
The issue is fixed. So closing the issue.

Thanks & Best Regards,
Rakesh

closed 2020-03-16 by rakeshm75