SymbiFlow/yosys #64: Branch : quicklogic : Dump report file for the area utilization

kkumar23, 2020-03-18:
[uart.zip](https://github.com/SymbiFlow/yosys/files/4349337/uart.zip)

Is there any option to dump the summary of the area utilization , as shown below:

***************************************************************
Device Utilization for QLAL4S3/PU90
***************************************************************
Resource                          Used    Avail   Utilization
---------------------------------------------------------------
IOs                               17      32        53.13%
LCs                               28      891       3.14%
DFFs                              47     891       5.27%
ClockCells                       2       5         40.00%
RAM8Ks                         0       8           0.00%
---------------------------------------------------------------

If not, supported can be provided ?

kgugala, 2020-03-18:
Yosys does not really have info about available resources. That kind of report should rather be generated in VPR.

kkumar23, 2020-03-18:
Thank  you for the quick reply. What is the option to dump the console log ?

kgugala, 2020-03-18:
when you run the build the flow should write the logs from each stage into files. So in the `pack.log` file you can find:

```
Pb types usage...
  PB-SYN_VCC   : 1
  PB-SYN_GND   : 1
  outpad       : 4
  VCC          : 1
  INPUT        : 1
  bidir_buf    : 5
  OUTPUT       : 4
  inpad        : 1
  BIDIR        : 5
  LOGIC        : 38
  PB-LOGIC     : 38
  f_frag       : 1
  GND          : 1
  q_frag       : 24
  c_frag       : 38
  PB-SYN_IO    : 5
	EMPTY: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
	PB-LOGIC: # blocks: 38, average # input + clock pins used: 16.8684, average # output pins used: 1.65789
	PB-ASSP: # blocks: 0, average # input + clock pins used: 0, average # output pins used: 0
	PB-SYN_IO: # blocks: 5, average # input + clock pins used: 2.8, average # output pins used: 0.2
	PB-SYN_VCC: # blocks: 1, average # input + clock pins used: 0, average # output pins used: 1
	PB-SYN_GND: # blocks: 1, average # input + clock pins used: 0, average # output pins used: 1
Absorbed logical nets 5 out of 71 nets, 66 nets not absorbed.
FPGA sized to 35 x 33 (ql-eos-s3)
Device Utilization: 0.04 (target 1.00)
	Block Utilization: 0.04 Type: PB-LOGIC
	Block Utilization: 0.16 Type: PB-SYN_IO
	Block Utilization: 1.00 Type: PB-SYN_VCC
	Block Utilization: 1.00 Type: PB-SYN_GND
```

The above example is for a simple 24 bit counter design. The `Pb types usage ...` counts usage of every block type used in the device. The `Device Utilization` section sums that up.

closed 2020-04-08 by kgugala