SymbiFlow/yosys #65: Branch: Quicklogic : Multiplier primitive instance should be qlal4s3_mult_cell_macro

rakeshm75, 2020-03-23:
Multiplier primitive instance should be qlal4s3_mult_cell_macro, so the black box instance should be the following in cells_sim.v file:

(* blackbox *)
module qlal4s3_mult_cell_macro (
	input [31:0] Amult,
	input [31:0] Bmult,
	input [1:0] Valid_mult,
	input sel_mul_32x32,
	output [63:0] Cmult );

endmodule 

User should be configuring the multiplier into x32 or x16 and should be connecting the signals appropriately.

Attached the primitive files ( cells_sim.v  & cells_map.v) , design example (rtl) and EDIF file for reference.
[Multipliers.zip](https://github.com/SymbiFlow/yosys/files/4367424/Multipliers.zip)

glatosinski, 2020-04-01:
The multiplier is added to Yosys and will soon be released on Github

glatosinski, 2020-04-08:
The multiplier is released

closed 2020-04-08 by glatosinski

kgugala, 2020-04-10:
@rakeshm75 all the mults macros are also supported in arch-defs

rakeshm75, 2020-04-13:
@kgugala, I have pulled the latest code. When I run the design with multiplier, I get the following error:

25.2.2. Analyzing design hierarchy..
ERROR: Module `\qlal4s3_mult_cell_macro' referenced in module `\AL4S3B_FPGA_IP' in cell `\u_qlal4s3_mult_cell_macro' is not part of the design.
make[3]: *** [quicklogic/tests/quicklogic_testsuite/design3/design3-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_synth.json] Error 1
make[2]: *** [quicklogic/tests/quicklogic_testsuite/design3/CMakeFiles/file_quicklogic_tests_quicklogic_testsuite_design3_design3-ql-chandalar_ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp_top.eblif.dir/all] Error 2
make[1]: *** [quicklogic/tests/quicklogic_testsuite/design3/CMakeFiles/design3-ql-chandalar_jlink.dir/rule] Error 2
make: *** [quicklogic/tests/quicklogic_testsuite/design3/CMakeFiles/design3-ql-chandalar_jlink.dir/rule] Error 2
[design3.zip](https://github.com/SymbiFlow/yosys/files/4469760/design3.zip)

kgugala, 2020-04-14:
hi @rakeshm75 I added your test to arch-defs repository https://github.com/antmicro/symbiflow-arch-defs/commit/7670933dfefc9237b050ad1fe231f5b38810f1ac

I ran it and it passes to the bitstream. It looks like you don't have the latest Yosys code. Support for the `qlal4s3_mult_cell_macro` cell has been added in this commit https://github.com/antmicro/yosys/commit/8d691b63910b2e81506c2201d496aaa41ea47577

There is one more issue with the design you attached - it get almost entirely optimized out during synthesis.

rakeshm75, 2020-04-14:
Hi Karol,

Its the same design which I had filed issue with Yosys that it optimizes the design completely.
The issue number #63 (https://github.com/SymbiFlow/yosys/issues/63). It was fixed and I could run the design on Yosys, create edif and run through SpDE tool. I have attached the EDF file created by Yosys and Precision for your reference.

We should have only qlal4s3_mult_cell_macro multiplier primitive not qlal4s3_mult_32x32_cell nor qlal4s3_mult_16x16_cell. 32x32 or 16x16 multiplier should be built from the base qlal4s3_mult_cell_macro.

(* blackbox *)
module qlal4s3_mult_cell_macro (
	input [31:0] Amult,
	input [31:0] Bmult,
	input [1:0] Valid_mult,
	input sel_mul_32x32,
	output [63:0] Cmult );
endmodule

[design3_EDF.zip](https://github.com/SymbiFlow/yosys/files/4475158/design3_EDF.zip)