SymbiFlow/yosys #68: Branch : Quicklogic : In symbiflow 'dffpc' not supported

rakeshm75, 2020-04-02:
When run the attached design through the symbiflow, we get the following error:

Error 1:
Type: Blif file
File: /home/rakeshm/symbiflow-arch-defs/build/quicklogic/tests/quicklogic_testsuite/design1/design1-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.eblif
Line: 9821
Message: Failed to find matching architecture model for 'dffpc'

# Load circuit
# Load circuit took 0.08 seconds (max_rss 10.4 MiB, delta_rss +4.3 MiB)
The entire flow of VPR took 0.11 seconds (max_rss 10.4 MiB)
make[3]: *** [quicklogic/tests/quicklogic_testsuite/design1/design1-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.net] Error 1
make[2]: *** [quicklogic/tests/quicklogic_testsuite/design1/CMakeFiles/file_quicklogic_tests_quicklogic_testsuite_design1_design1-ql-chandalar_ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp_top.net.dir/all] Error 2
make[1]: *** [quicklogic/tests/quicklogic_testsuite/design1/CMakeFiles/design1-ql-chandalar_jlink.dir/rule] Error 2
make: *** [quicklogic/tests/quicklogic_testsuite/design1/CMakeFiles/design1-ql-chandalar_jlink.dir/rule] Error 2
[rtl.zip](https://github.com/SymbiFlow/yosys/files/4422364/rtl.zip)

kgugala, 2020-04-03:
This is fixed now. Please fetch the latest code.

rakeshm75, 2020-04-03:
Hi Karol,

With the latest code, compilation of any design is failing. I get the following error:

Design 1:

3.28.9. Finished OPT passes. (There is nothing left to do.)
ERROR: Can't load module `./ql-iob': /home/rakeshm/antmicro_install/bin/../share/yosys/plugins/ql-iob.so: cannot open shared object file: No such file or directory
make[3]: *** [quicklogic/tests/quicklogic_testsuite/clock_tree_design/clock-tree-design-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_synth.json] Error 1
make[2]: *** [quicklogic/tests/quicklogic_testsuite/clock_tree_design/CMakeFiles/file_quicklogic_tests_quicklogic_testsuite_clock_tree_design_clock-tree-design-ql-chandalar_ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp_top.eblif.dir/all] Error 2
make[1]: *** [quicklogic/tests/quicklogic_testsuite/clock_tree_design/CMakeFiles/clock-tree-design-ql-chandalar_jlink.dir/rule] Error 2
make: *** [quicklogic/tests/quicklogic_testsuite/clock_tree_design/CMakeFiles/clock-tree-design-ql-chandalar_jlink.dir/rule] Error 2

Design 2:

4.28.9. Finished OPT passes. (There is nothing left to do.)
ERROR: Can't load module `./ql-iob': /home/rakeshm/antmicro_install/bin/../share/yosys/plugins/ql-iob.so: cannot open shared object file: No such file or directory
make[3]: *** [quicklogic/tests/quicklogic_testsuite/camif/camif-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_synth.json] Error 1
make[2]: *** [quicklogic/tests/quicklogic_testsuite/camif/CMakeFiles/file_quicklogic_tests_quicklogic_testsuite_camif_camif-ql-chandalar_ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp_top.eblif.dir/all] Error 2
make[1]: *** [quicklogic/tests/quicklogic_testsuite/camif/CMakeFiles/camif-ql-chandalar_jlink.dir/rule] Error 2
make: *** [quicklogic/tests/quicklogic_testsuite/camif/CMakeFiles/camif-ql-chandalar_jlink.dir/rule] Error 2

[log_camif.zip](https://github.com/SymbiFlow/yosys/files/4428011/log_camif.zip)
[log_clock_tree_design.zip](https://github.com/SymbiFlow/yosys/files/4428013/log_clock_tree_design.zip)

kgugala, 2020-04-03:
hi @rakeshm75 
the plugin is required to handle the `SFBIO` pins. Please refer to the readme file https://github.com/antmicro/symbiflow-arch-defs/blob/quicklogic-upstream-rebase/quicklogic/README.md

It has been updated with instructions for building and installing the plugin

kgugala, 2020-04-08:
@rakeshm75 did it work for you? If yes, please close the issue

rakeshm75, 2020-04-08:
Hi Karol,

   Yes, this issue is resolved, so closing the issue.

Thanks & Regards,
Rakesh

closed 2020-04-08 by rakeshm75