SymbiFlow/yosys #69: Branch : Quicklogic : In symbiflow ''ram8k_2x1_cell_macro'' not supported

rakeshm75, 2020-04-02:
When run the attached design through the symbiflow, we get the following error:

Error 1: 
Type: Blif file
File: /home/rakeshm/symbiflow-arch-defs/build/quicklogic/tests/quicklogic_testsuite/design1/design1-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top.eblif
Line: 11551
Message: Failed to find matching architecture model for 'ram8k_2x1_cell_macro'
[rtl1.zip](https://github.com/SymbiFlow/yosys/files/4422383/rtl1.zip)

kgugala, 2020-04-09:
this should be fixed now. Please fetch the latest code and run the tests again. There is also a RAM test included in the repository https://github.com/antmicro/symbiflow-arch-defs/tree/quicklogic-upstream-rebase/quicklogic/tests/ram

rakeshm75, 2020-04-13:
This issue is fixed, so closing the issue.

closed 2020-04-13 by rakeshm75