SymbiFlow/yosys #70: Branch : Quicklogic : In symbiflow, AssertionError reported

rakeshm75, 2020-04-06:
In symbiflow, AssertionError reported:

make camif-ql-chandalar_jlink

Following error reported:
AssertionError: ('gpmc_ad(0)', 'FBIO_20')
make[3]: *** [quicklogic/tests/quicklogic_testsuite/camif/camif-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_iomux.jlink] Error 1
make[3]: *** Deleting file `quicklogic/tests/quicklogic_testsuite/camif/camif-ql-chandalar/ql-s3-ql-eos-s3-virt-ql-eos-s3-wlcsp/top_iomux.jlink'
make[2]: *** [quicklogic/tests/quicklogic_testsuite/camif/CMakeFiles/camif-ql-chandalar_jlink.dir/all] Error 2
make[1]: *** [quicklogic/tests/quicklogic_testsuite/camif/CMakeFiles/camif-ql-chandalar_jlink.dir/rule] Error 2
make: *** [quicklogic/tests/quicklogic_testsuite/camif/CMakeFiles/camif-ql-chandalar_jlink.dir/rule] Error 2
[rtl.zip](https://github.com/SymbiFlow/yosys/files/4437703/rtl.zip)

kgugala, 2020-04-08:
this is an assert in the jlink script generator. The bitstream itself is generated correctly. We'll take a look on the script

kgugala, 2020-04-08:
@rakeshm75 this is now fixed with https://github.com/antmicro/symbiflow-arch-defs/commit/c15f616aa9284e4ae4ed6d8de84af741063942ed

tpagarani, 2020-04-08:
@kgugala , Thanks for providing this fix. While reviewing this script, we noticed that you are using the "mode" setting to set bit 5 and bit 11 on Pad config register. Actually for IOs used by FPGA, there is no need to set bit 5 and bit 11. They are only applicable when IO is not used by FPGA. In case of FPGA, OEN and REN are directly routed to FPGA interface. @rakeshm75  will send you a document to explain this. So you can skip setting these bits if ctrl_sel is Fabric.

rakeshm75, 2020-04-09:
This issue is resolved, so closing the issue.

closed 2020-04-09 by rakeshm75